NxRLab/ModernRobotics #4: Maximum recursion limit of 500 reached

ASANIART, 2018-05-06:
When I run any of the Matlab codes (with the given examples on the code) I get this message:

"Maximum recursion limit of 500 reached. Use set(0,'RecursionLimit',N) to change the limit. Be aware that
exceeding your available stack space can crash MATLAB and/or your computer."

jarvisschultz, 2018-06-01:
This doesn't happen on any MATLAB versions that we've tested the library on. This likely points to an issue with your MATLAB installation. Do you have a simple example of code that produces the error? Saying that "any of the Matlab codes" causes the error doesn't give us much to go off of. A small snippet that produces the error might help us narrow down what the issue is.

jarvisschultz, 2018-06-21:
Closing due to lack of activity.

closed 2018-06-21 by jarvisschultz