NxRLab/ModernRobotics #8: Formatting issues in Python library docstrings

jarvisschultz, 2018-11-14:
There are a variety of formatting issues in the docstrings in the Python library. While this isn't a huge deal, it does mean that some of the example usage cannot be directly copy-pasted into a Python interpreter and have it run successfully. This is true even with an indention-aware, "smart" interpreter like IPython's `%paste` function. Here are a few issues that I noticed:

+ Inconsistent indentation on some of the sample code. For example, see [ForwardDynamicsTrajectory](https://github.com/NxRLab/ModernRobotics/blob/77dcf6617c353b1fd009aa1c10762b00b3205eaf/packages/Python/modern_robotics/core.py#L1412-L1429).
+ Mixed usage of tabs and spaces for indentation, even within a single docstring. Python not only [discourages use of tabs](https://github.com/NxRLab/ModernRobotics/blob/77dcf6617c353b1fd009aa1c10762b00b3205eaf/packages/Python/modern_robotics/core.py#L1412-L1429) for indentation, but on Python 3 they've actually disallowed mixed usage. To see this on GitHub, it might help to [change the spacing](https://stackoverflow.com/questions/8833953/how-to-change-tab-size-on-github/23522945#23522945) that GitHub renders tabs with. Here's [one example](https://github.com/NxRLab/ModernRobotics/blob/77dcf6617c353b1fd009aa1c10762b00b3205eaf/packages/Python/modern_robotics/core.py?ts=2#L1470-L1486) with the tab width set to 2 spaces. Notice the inconsistent indentation?
+ Some of the more in-depth examples explicitly import `modern_robotics` but they fail to import `numpy` or `print_function`.
+ There are a few spelling issues [example](https://github.com/NxRLab/ModernRobotics/blob/77dcf6617c353b1fd009aa1c10762b00b3205eaf/packages/Python/modern_robotics/core.py#L185)
+ Some trailing whitespace (not a big deal).

zrwang, 2018-12-07:
I should look into your pull request before creating one. I found some similar problems mentioned in this issue. I also fixed wrong indentations, especially in the function description comments. Anyway, thanks for your effort.

jarvisschultz, 2018-12-07:
@zrwang thanks for your attention to detail!

@HuanWeng can you please review both of these PRs, and accept whichever is more appropriate and then close the other?

jarvisschultz, 2018-12-07:
Now that #9 is merged, we can close this issue.

closed 2018-12-07 by jarvisschultz