NxRLab/ModernRobotics #10: C++ library port?

Le0nX, 2018-11-15:
Good morning! I'm one of your students of Modern Robotics MOOC on Coursra and writing you with a question from Russia! 

As I can see there are only Mathematica, Matlab and Python packages for this course. I can write a C++ one and make a pull request to your github repo. How do you look at it?

jarvisschultz, 2018-11-20:
This is a great offer, and a C++ implementation of this library would be a very useful contribution. That said, we don't necessarily want to commit to maintaining code that we didn't write and that is outside the original scope of this software library. @Le0nX and I discussed this via email, and the outcome is that he is going to independently develop the C++ library and push it to his own repo, and once it is ready we will feature it in documentation on the book's homepage and in this repository.

Looking forward to seeing the implementation!

closed 2018-11-20 by jarvisschultz

stevenjj, 2018-11-20:
Thanks for doing this @Le0nX . My labmate actually started writing parts of it in C++ as part of our research project together. But we only wrote the functions we needed to help compute the spatial Jacobian and forward kinematics. But, I believe it may still be of use to you.

This is the header file https://github.com/stevenjj/dreamer/blob/master/Sandboxes/gaze_control/src/gaze_control_cpp/modernRobotics.h

and this is the c++ implementation file: https://github.com/stevenjj/dreamer/blob/master/Sandboxes/gaze_control/src/gaze_control_cpp/modernRobotics.cpp

Le0nX, 2018-11-21:
Hi, @stevenjj ! Thanks for your reply! Your labmate's code samples will be very helpful. Are you going to use the entire library in the future? Do you have any special wishes on build system or structure? 
At this moment we are going to use Cmake and Eigen.