NxRLab/ModernRobotics #16: Error in ProjectToSO3(mat) (Python) function *** CHAPTER 3: RIGID-BODY MOTIONS ***

DjoleMNE, 2019-02-08:
In Chapter 3, exercise 3.10 and in Appendix E (Linear Algebra Review, section E.3.2), the problem of a 3x3 matrix "escaping" the SO(3) manifold due to numerical roundoff errors, is presented. For the case when a given matrix is close to the SO(3), the solution for projecting the matrix to the closest matrix in SO(3) using singular-value decomposition, is defined. However, for the case when the matrix is not sufficiently close to SO(3), i.e. the determinant of resulting (SVD processed) matrix is < 0, the solution is not described theoretically, but it has been coded in the ProjectToSO3(mat) (Python) function. More specifically:
```
   if np.linalg.det(R) < 0:
    # In this case the result may be far from mat.
        R[:, s[2, 2]] = -R[:, s[2, 2]]
    return R
```
Nevertheless, the given solution code is falling and the error is the following:

```
---> 26         R[:, s[2, 2]] = -R[:, s[2, 2]]
     27     return R

IndexError: too many indices for array
```

Is it possible to share the reference/s (article, book), where a theoretical solution is provided for this special case, so readers of "CHAPTER 3: RIGID-BODY MOTIONS" and users of this function have better theoretical background for the aforementioned coded solution, and maybe implement the fix for the above error in the code?

HuanWeng, 2019-02-09:
Hi Djordje,
Thank you for pointing this out! This is a bug and should be
```
R[:, 2] = -R[:, 2]
```
Section E.3.2 of [this file](http://hades.mech.northwestern.edu/images/e/e2/AppendixE-linear-algebra-review-Sept21-2018.pdf) was our reference. A paper called *Least-Squares Estimation of Transformation Parameters Between Two Point Patterns* is also quite useful.
I will fix this and add the reference to this function soon. Let me know if you have other questions.
Many thanks,
Huan Weng

DjoleMNE, 2019-02-09:
Hi Huan,

thank you for the solution and references!

Best regards,
Djordje Vukcevic

closed 2019-02-09 by DjoleMNE