NxRLab/ModernRobotics #18: Library translation to Julia

ferrolho, 2019-02-23:
Similar to #10, I have translated the library to [Julia](https://julialang.org/). Documentation needs a bit of improvement, but all the functionality should be there (only methods missing are `ScrewTrajectory` and `CartesianTrajectory`, which I'll add soon enough anyway).

A _Jupyter Notebook_ with the authors' original examples: [Modern Robotics Examples](https://github.com/ferrolho/ModernRobotics.jl/blob/e754a8d9d8c537ea6b710af77ed5a8a198e7c9e4/examples/Modern%20Robotics%20Examples.ipynb)

Ongoing Julia package PR: [JuliaLang/METADATA.jl/pull/21669](https://github.com/JuliaLang/METADATA.jl/pull/21669)

Repository: [ferrolho/ModernRobotics.jl](https://github.com/ferrolho/ModernRobotics.jl)
Release: [v0.1.0](https://github.com/ferrolho/ModernRobotics.jl/releases/tag/v0.1.0)

paulyang1990, 2019-02-23:
I know nothing about Julia but good luck!

closed 2019-03-03 by ferrolho

HuanWeng, 2019-03-04:
I will refer your work as I did for C++ version in the main README. Hope that helps.