NxRLab/ModernRobotics #20: Licensing Information Missing

cbonitz, 2019-03-03:
Thank you for publishing this library.

Sadly, the repository does not contain any license statement, which is common practice on GitHub.

The [PyPI page](https://pypi.org/project/modern_robotics/) of the package mentions the MIT license. It would be great if you could add a LICENSE.txt containing the [license](https://opensource.org/licenses/MIT), with an added copyright statement, to this repo. The only thing you would see also [Licensing a repository](https://help.github.com/en/articles/licensing-a-repository) by GitHub.

HuanWeng, 2019-03-26:
Thank you for reminding us! We have added it.

cbonitz, 2019-03-26:
Thank you for adding the license.

closed 2019-03-26 by cbonitz