NxRLab/ModernRobotics #21: Can the maintenance team look into the C++ library and officially endorse it

paulyang1990, 2019-03-26:
Hi, 

A group of people including me has ported your library into C++ (https://github.com/Le0nX/ModernRoboticsCpp). I am using it now almost every day on my research projects. I am wonder can you verify this C++ version and merge it into your official repo? If not, please tell us what need to be further refined so that you can endorse it. So in the future if you have modifications to the official repo we can quickly update C++ version too.

HuanWeng, 2019-06-10:
Hi Shuo,

Sorry for the late reply. Thank you and other contributors working on that version! I understand it is great to keep this C++ version up to date, but for now we won't merge it. 

This library accompanies the textbook so there won't be too many updates. And merging it to the official repo means we should take the responsibility of maintaing it and we need to have someone look over this version before merging, which we don't have that manpower to do now. 

We have had the link directing to this version in our README.

Many thanks

paulyang1990, 2019-07-22:
Hi Huan, 

Thanks for your reply! Totally understood. We will keep maintaining the C++ version.

closed 2019-07-22 by paulyang1990