NxRLab/ModernRobotics #32: The RNE

priscillascu, 2020-11-26:


priscillascu, 2020-11-26:
Hi, i'm using the RNE inverse dynamics algorithm function InverseDynamics in Matlab 2019b. But the torques that i use this function got isn't equal to which I use SimScape measured. And when i using the calculated torque as input torque of my SimScape input, the robot arm moves wrong.

Please tell me how to use this function correctly. THANKS.

HuanWeng, 2020-11-26:
Hi Haixin,

It's hard for me to believe the function InverseDynamics is wrong since it has been used by hundreds of people and no similar issues reported. I guess the inconsistency between the results from our library and your simulation was caused by other issues (Wrong inputs? Should use InverseDynamicsTrajectory instead?, etc.), but I'm happy to work with you to fix the bug if you really find any.

The description and usage example of each function are at the beginning of the function definition, and can be checked by running "help FUNCTION_NAME". The algorithms used are all mentioned in the corresponding chapters in our textbook.

Good luck

priscillascu, 2020-11-27:
Thanks for your help. I'll check it.😃😃😃😃😃

priscillascu, 2020-11-29:
Hi, I have checked and corrected some model parameters, and use InverseDynamicsTrajectory instead. It works.

But when all joints are stationary, there is a small error between the results from your library and my simulation. This causes my model to move well at the beginning, but it will crash after a while. I have no idea about this problem.

Hope you guys can help me.

The torque errors and joint moving are as follows.
[image]
The joint moving
[image]