NxRLab/ModernRobotics #34: InverseDynamics and ForwardDynamics results do not match

priscillascu, 2020-12-07:
Hi guys. I encountered a strange problem.
When I use the torques which calculated by InverseDynamics function as inputs of ForwardDynamics function. Then I get a bad results of angles and angular velocity. 
Both of them are based on the same model, and the input torques of ForwardDynamics are equal to InverseDynamics outputs. And the integration algorithm I used is EulerStep function.
Hope you guys can help me.

priscillascu, 2020-12-08:
[image]
Here is the joint angle. The joint torque is the same.

humeiling1122, 2020-12-23:
Sorry,I can't check your picture.You mean the inversedynamics' results or the inversedynamicsTrajectory's results?