NxRLab/ModernRobotics #47: Torque calculation problem with InverseDynamicsTrajectory function using UR10e URDF

sm3304love, 2022-10-31:
Hello. I'm going to implement the UR10e  Force control using InverseDynamicsTrajectory Function. However, I would like to ask you a question because the calculated torque does not seem to be the correct value.

Here is UR10e Dynamcis parameter

Slist :
[image]

Mlist : 
[image]
[image]

Glist :
[image]
[image]

And, this is my code
```
%init pose to end pose torque compute

thetastart = [0; 0; 0; 0; 0; 0];
thetaend = [pi/2;0; 0;0;0;0];
% Total time of the motion in seconds from rest to rest
Tf = 5;
%The number of points N > 1 (Start and stop) in the discrete representation of the trajectory
dt = 0.01;
%Create a trajectory to follow

N = Tf / dt;
%he time-scaling method, where 3 indicates cubic third-order polynomial) time scaling and 5 indicates quintic (fifth-order polynomial) time scaling.
method = 5;

traj = JointTrajectory(thetastart, thetaend, Tf, N, method);

thetamat = traj;

dthetamat = zeros(1000, 6);

ddthetamat = zeros(1000, 6);

dt = Tf / (N - 1);
for i = 1: N - 1
  dthetamat(i + 1, :) = (thetamat(i + 1, :) - thetamat(i, :)) / dt;
  ddthetamat(i + 1, :) = (dthetamat(i + 1, :) - dthetamat(i, :)) / dt;
end
g = [0; 0; -9.8];
Ftipmat = zeros(N, 6); 

taumat = InverseDynamicsTrajectory(thetamat, dthetamat, ddthetamat, ...
                                 g, Ftipmat, Mlist, Glist, Slist)
%Output using matplotlib to plot the joint forces/torques
time=0: dt: Tf;
plot(time, taumat(:, 1), 'b')
hold on
plot(time, taumat(:, 2), 'g')
plot(time, taumat(:, 3), 'r')
plot(time, taumat(:, 4), 'y')
plot(time, taumat(:, 5), 'k')
plot(time, taumat(:, 6), 'm')
title('Plot for Torque Trajectories')
xlabel('Time')
ylabel('Torque')
legend('Tau1', 'Tau2', 'Tau3','Tau4','Tau5','Tau6')
```

Taumat calculated by the above code  
```
         0  -71.0770  -23.0659         0   -0.0000         0
    0.0099  -71.0780  -23.0660   -0.0000    0.0000         0
    0.0592  -71.0828  -23.0665   -0.0003    0.0001         0
    0.1177  -71.0886  -23.0670   -0.0005    0.0002         0
[... 474 lines cut ...]
   -1.1429  -70.9650  -23.0549    0.0049   -0.0018         0
   -1.0981  -70.9694  -23.0553    0.0047   -0.0017         0
   -1.0526  -70.9738  -23.0557    0.0046   -0.0016         0
   -1.0064  -70.9783  -23.0561    0.0044   -0.0016         0
   -0.9596  -70.9829  -23.0566    0.0042   -0.0015         0
   -0.9120  -70.9875  -23.0570    0.0040   -0.0014         0
   -0.8639  -70.9922  -23.0575    0.0038   -0.0013         0
   -0.8150  -70.9970  -23.0579    0.0035   -0.0013         0
   -0.7655  -71.0019  -23.0584    0.0033   -0.0012         0
   -0.7153  -71.0068  -23.0589    0.0031   -0.0011         0
   -0.6644  -71.0118  -23.0594    0.0029   -0.0010         0
   -0.6129  -71.0168  -23.0599    0.0027   -0.0010         0
   -0.5607  -71.0220  -23.0604    0.0024   -0.0009         0
   -0.5077  -71.0272  -23.0609    0.0022   -0.0008         0
   -0.4541  -71.0324  -23.0614    0.0020   -0.0007         0
   -0.3998  -71.0378  -23.0620    0.0017   -0.0006         0
   -0.3448  -71.0432  -23.0625    0.0015   -0.0005         0
   -0.2891  -71.0486  -23.0630    0.0013   -0.0005         0
   -0.2327  -71.0542  -23.0636    0.0010   -0.0004         0
   -0.1755  -71.0598  -23.0642    0.0008   -0.0003         0
   -0.1177  -71.0655  -23.0647    0.0005   -0.0002         0
   -0.0592  -71.0712  -23.0653    0.0003   -0.0001         0
```
Since only the first joint rotates by pi/2, I think the torque absolute value of the first joint should increase and then decrease.
However, the value of the calculated torque is really confusing because it has a value that goes between positive and negative values. can you please help me?

priscillascu, 2022-10-31:
收到，我会尽快给您答复。杨海鑫

sm3304love, 2022-10-31:
When I implemented that code in ROS melodic, the movement of UR10e in Gazebo looks like the following image.

[image]

sm3304love, 2022-10-31:
http://hades.mech.northwestern.edu/images/b/b6/UR5-parameters.m 

The same thing happened when I used the UR5 parameter provided by Modern Robotics - Northwestern Mechatronics Wiki!

HuanWeng, 2022-10-31:
The torque value going positive and negative actually makes sense to me since the arm speeds up (positive accel. required) and slows down (negative accel. required), as the motion trajectory. I see your .fig animation and that confused me because at least it seems like the arm didn't start at the home configuration. 

I would suggest checking out the Gazebo simulation, and then the kinematics using your trajectory, and then dynamics. 

I can take a look at that function if you still have trouble and have a complete example with all parameters in the code.

sm3304love, 2022-11-01:
@HuanWeng 

As you said, I used Gazebo Simulation to control the UR10e by giving the Effort command.
When the robot was summoned from Gazebo, it could not start from home configuration due to gravity, so I read the joint position data from Gazebo and put it in the initial theta value. The results are shown in the image below.

In my opinion, the calculated torque is too small to move the UR10e.

[image]

stevenjj, 2022-11-01:
@sm3304love : Instead of using `InverseDynamicsTrajectory`, I would first use `GravityForces(thetalist, g, Mlist, Glist, Slist)` with very small PD gains and see if you can hold a joint configuration.

Checking if gravity compensation works first is an easy way to isolate errors.

Also, at the zero configuration, you can easily calculate the expected torque value since you know the center-of-mass locations of all the links. So you will have another way to check for torque consistency.

sm3304love, 2022-11-04:
@stevenjj 
@HuanWeng 

As you said, I tried to apply it using GravityForces(thetalist, g, Mlist, Glist, Slist). But still, the UR10e does not resist gravity. If the Effort value is set arbitrarily high, the UR10e will move. Perhaps the torque required by the robot is very large.

[image]

But I know that torque -23.0659Nm is by no means a small value. In fact, if I add a torque of -71.077Nm to only the 2nd joint while keeping the torque of the other links at zero, the 2nd joint will move. However, if I give -23.0659Nm to only the 3rd joint, the 3rd joint will not move.
[image]
[image]

When I checked the coordinate system of UR10e, it was the same as the image below. I understand that the coordinate system of the manipulator generally sets the rotation axis of the motor to the Z axis. Can this be a problem?

[image]

If you want to reproduce the problem, I can send you the ROS package as an email with an explanation. Please help me!

sm3304love, 2022-11-09:
I think there was a problem with the Mlist. I don't think it's a problem with Gazebo or code.

closed 2022-11-09 by sm3304love