NxRLab/ModernRobotics #52: I have question about ComputedTorque function.

sm3304love, 2023-02-06:
[image]

In `ComputedTorque` function, Why term `h(θ, ˙θ)_hat` is calculated by `InverseDynamics(thetalist, dthetalist, ddthetalistd, g, zeros(6, 1), Mlist, Glist, Slist)` instead of using `VelQuadraticForce` and `GravityForces`?

HuanWeng, 2023-02-06:
In that function, $\tau$ is calculated as a sum of $\hat{M}(\theta)(\text{ACCELERATION-VALUE-FROM-PID-TERMS})$ and `InverseDynamics(thetalist, dthetalist, ddthetalistd, g, zeros(6, 1), Mlist, Glist, Slist)`, the later equals to $\hat{M}(\theta)\ddot{\theta_d}+\hat{h}(\theta,\dot{\theta})$.

sm3304love, 2023-02-07:
Thank you for your answer. I have more questions. In the book, the ComputedTorque function appears to implement the formula below. Why is the last part calculated with  $\hat{M}(\theta)\ddot{\theta_d}+\hat{h}(\theta,\dot{\theta})$ in the ComputedTorque function code?

[image]

closed 2023-02-20 by sm3304love