NxRLab/ModernRobotics #56: Matlab Functions FKinBody and FKinSpace Output Incorrect Values

Kilian-Olen, 2023-06-25:
I'm not sure how this has not been resolved yet but FKinBody and FKinSpace do not return the correct values. This has been reported by a few users on coursera and after looking into the functions, all each function needed was a simple fix.

Below is the original code for FKinBody.m (this reports no issues however the values are incorrect)
T = M;
for i = 1: size(thetalist)
    T = T * MatrixExp6(VecTose3(Blist(:, i) * thetalist(i)));
end
end

Below is my correction which does output the correct values
T = M;
for i = 1: length(thetalist)
    T = T * MatrixExp6(VecTose3(Blist(:, i) * thetalist(i)));
end
end

Notice in the for loop that the size function has been adjusted to the length function. This issue is also present in FKinSpace.m and the same fix works there to. Please update the Matlab library functions to prevent future headaches.

HuanWeng, 2025-10-08:
We don't think that's an error. Please check this thread:
https://github.com/NxRLab/ModernRobotics/issues/2#issuecomment-385415238

jryishen, 2025-11-20:
I understand that the matrix dimension is critical.  However, for the the perspective to enable the MATLAB script to run, using the "size" command cause error message,  

Error using :  
Colon operands must be real scalars.  

Error in  (line 26) 
for i = 1: size(thetalist)   
            ^

The main issue is the "for" loop in MATLAB cannot take a 1x2 matrix returned by "size".  Revise it to "length" will return an 1x1 scalar.  Thus the FKinBody and FKinSpace can run.  Is it an error or not by using "size"?

priscillascu, 2025-11-20:
收到，我会尽快给您答复。杨海鑫